witnessmenow/arduino-youtube-api #5: List of subscribers (request)

marcfon, 2017-04-11:
Would it be possible to add a the option to get the list of subscribers? The YouTube API seems to support the function.
https://developers.google.com/youtube/v3/docs/subscriptions/list

witnessmenow, 2017-04-12:
Looks like it's possible ok. I am swamped for the next few weeks with other work but if somebody wanted to add it it shouldnt be too difficult.

First step would be to test the API on your computer to figure out how it works and make sure it's bringing back the data you want.

Second would be to modify the library, I would add another method [like getChannelStatistics](https://github.com/witnessmenow/arduino-youtube-api/blob/master/src/YoutubeApi.cpp#L86). Build up the url the same way that worked for you in the first step. The Json part will need to be changed as well to match the response from the API

That would be pretty much it. Sorry that I cant take it up at the moment

joeybab3, 2017-04-15:
Are you looking to get newly subscribed people? Because I  feel like the list would quickly become too much for the arduino to handle especially for people with 1k+ subs

marcfon, 2017-04-17:
@witnessmenow I looked in the current code and it looks fairly easy to add the new function. I also played around with the YouTube API but got stuck on that one. I can't seem to return the subscribers in the order I want. Will need to do some more testing.

@joeybab3 Yeah I'm looking to get the latest subscribers. Not all of them. Can't figure out the right YouTube API call though. It seems to return the subscribers in random order instead of the date subscribed.

marcfon, 2017-04-17:
I can't get it to work using the API key. I've managed to get it working using the OAuth access token but it takes a few additional steps to set this up. Would be nice if the API key would work.

marcfon, 2017-04-22:
So it turned out to be quite a project to implement the recent subscribers function. The result is that I've forked the project and implemented **OAuth2 support** in the library. This basically opens up the way to support the whole YouTube API. I'm sure the my code can be further optimised but the basics work. You can find the code at https://github.com/marcfon/arduino-youtube-api

My fork breaks the (backwards) compatibility with the current code as I've fully removed the dependancy on the API key. So it's up to you how you'd like to use it.