witnessmenow/arduino-youtube-api #10: Channel Stats not printing on Serial Monitor

HeathenHacks, 2019-01-24:
I'm not sure if I am the only one having this.

I'm using the ESP8266 -> ChannelStatistics code example. Followed all the instructions. 

Connecting Wifi: 
15:49:46.362 -> ..........
15:49:52.115 -> WiFi connected
15:49:52.115 -> IP address: 
15:49:52.115 ->

Shows up fine, but my channel's stats does not show up, no matter how long I wait.

witnessmenow, 2019-01-24:
What version of esp8266 core are you using?

If it's 2.5 , you probably need to add client.setInsecure(); to your setup
method

jus-tin, 2019-01-24:
Oh, thanks for the fast response.

I'm using version 2.4.2

jus-tin, 2019-01-24:
Here's a screenshot.

[image]

witnessmenow, 2019-01-24:
Do you get a valid IP address in the serial monitor?

jus-tin, 2019-01-24:
Yes. I'm also using a generic D1 Mini and I used the WEMOS D1 Mini board, if that could help.

jus-tin, 2019-01-25:
Fixed it. Sorry for the hassle, sir. 

My bad.

JaZzMaN251, 2019-01-26:
> Fixed it. Sorry for the hassle, sir.
> 
> My bad.

How did you fix it? I'm having the same issue.

Maximuskembo, 2019-02-03:
How did you fix this? I have the same problem. Only wifi connection and ip adress show up in serial monitor. Please help :)

JaZzMaN251, 2019-02-03:
I ended up creating a new project on Google API and creating a new YouTube API key. After a couple minutes, the key was activated and it showed the stats. 
(just creating a new API key in the existing project didn’t work, I really needed to create a new project)

Maximuskembo, 2019-02-03:
Thanks for you answer.
I tried to create a new project in Google API and inserted the new API key (with no restrictions). 
I have waited for 30 minutes now and no difference from the serial monitor. See screenshot.
[image]
Do you know if there is any way of monitoring if the ESP8266 sends a request to google? It looks like there is no activety?

JaZzMaN251, 2019-02-03:
You can add a couple messages like this in between the code to print to the monitor:

Serial.println("start stat request");

Maximuskembo, 2019-02-03:
Ok, I will try to do that.
I'l let you know how it goes :)
Thanks

witnessmenow, 2019-02-03:
Are you using ESP8266 2.5 beta? if so you will need to add this line to
your setup method:

client.setInsecure();

Maximuskembo, 2019-02-04:
Hi

Thank you so much, adding that line to the setup right here: 
[image]
 Fixed the problem. I am using the Wemos D1 mini pro board.
I am making a youtube video about making this live counter so I was wondering if its ok to include this thread in the video?
Thanks again for helping out!!

regards 
Willys Garage Norway

closed 2019-02-18 by HeathenHacks