witnessmenow/arduino-youtube-api #13: Library unable to connect to YouTube APIs

oxivanisher, 2019-02-20:
I am also trying to implement a YouTube subscriber counter. As basis, I used an [existing project](https://github.com/SensorsIot/Youtube-Subscriber-Counter/tree/master/YoutubeCounterNeopixelV1.0) (not mine) which seemed to work just fine. But my subscriber numbers just always kept being 0. I debugged as far as possible and it seems, that the connection to www.googleapis.com:443 are just not working. If I open the same URL as generated in this library for example in my browser, I get the numbers as expected.
The only thing I am seeing is:
```
Connecting to: 172.217.168.74 - 443      << my debug output
pm open,type:2 0
```
Since it never reaches `.... connected to server`, the problem has to be with the line:
`if (client->connect(YTAPI_HOST, YTAPI_SSL_PORT)) {`
I have no idea how to debug this further. Is it possible that google changed something (SSL Cert or such) ?
Any other pointers or ideas?

The other project basically uses the example code from this library here. Since the NTP time-fetch works, the Network connectivity of the Weemos D1 Mini is not the problem. Also replacing the URL with the IP has the same result.

witnessmenow, 2019-02-20:
The beta release of esp8266 (2.5) has made changes around connecting, and
I'm guess that's what you have installed.

It will no longer connect without a fingerprint or for you to specifically
say I don't want to use a fingerprint.

You can either downgrade the board library to 2.4 or

In your setup you can either

client.setFingerprint("ae 12 cd ......");
(But I believe YouTube change their fingerprint very often)

Or

client.setInsecure();
(I'd just do it this way)

oxivanisher, 2019-02-20:
Hey @witnessmenow, thank you very much for your fast answer!

You put me on the right track, even if I don't understand why it "works" now:
I am using Platform.IO (Atom). The espressif8266 was/is version 2.4.? (stable). But then I tried the dev. Version, which just got bumped to 2.5 days before. It at first did not work even with the `client.setInsecure();` added.
But now it gets really confusing. If I add the build-flag `DEBUG_ESP_CORE`, it works (also on the older lib version). 🤷‍♂️

For me it looks like a bug in the espressif8266 lib. I am happy that I found a way to go forward with the project (still a lot of soldering ahead) and will probably revisit the code once 2.5 is stable. For anyone finding this, I added

```
[env]
.....
build_flags =
  -DDEBUG_ESP_CORE
```

to my `platformio.ini`.

Thanks again to @witnessmenow, the issue can IMHO be closed.

closed 2019-02-20 by oxivanisher

oxivanisher, 2019-03-05:
It was definitely a bug in the platform.io implementation of the Arduino framework for the weemos d1 minis. If anybody else has that problem, please update the espressif platform to the current version (and use `client.setInsecure();`.