witnessmenow/arduino-youtube-api #18: Not working anymore?

necoinside, 2020-03-05:
Hi,
I'm trying with wemos d1 r1, I've api_key that works with link that written in ReadMe, also board connects wifi and able to ping google.
But nothing more, no return as json to the board.
Note: Already downgraded ArduinoJson to 5.x and also tried to downgrade esp8266 lib to 2.5.0 but with no result.
Note2: client.setInsecure(); line added to setup() place too and again with no result.

rsuriano, 2020-03-12:
I think it might be a problem with the API itself. I had a project using this library and Youtube Statistics and it suddenly stopped working, same as you're describing. Could you come up with a fix?

defwheezer, 2020-04-01:
I also had the same with youtubeapi problem, but it worked after I added the following line in setup():

client.setInsecure(); 

-Also using ESP8266

necoinside, 2020-04-02:
I rewrote the sketch including BearSSLClient, its requirements and also client.setInsecure(); than it has started to work.

ssttaarr33, 2020-04-05:
Can you please share the sketch? I can't get it working

necoinside, 2020-04-05:
I've cleanly modified the original sketch for you and uploaded here @ssttaarr33 
[ChannelStatisticsModified.zip](https://github.com/witnessmenow/arduino-youtube-api/files/4433744/ChannelStatisticsModified.zip)

closed 2020-04-05 by necoinside