witnessmenow/arduino-youtube-api #19: Youtube API not working

NoName-arduino, 2020-05-13:
**Hey**,
I know its been a long time since the last post, but i have a problem.
So i have this code (see attachment) and i noticed problems when using static ip adresses (in the code config and on the router side.
Help is much appreciated.
i get no error messages but "api.getChannelStatistics(CHANNEL_ID)" returns always false.

API token and url works fine
[ex.zip](https://github.com/witnessmenow/arduino-youtube-api/files/4620955/ex.zip)

(Sorry for my bad english)

AntoniousUA, 2020-05-15:
Have same issue. Made _debug (in YoutubeApi.h ) as true and see in Serial monitor that body have all data, but parsing do not working.

`
17:06:19.195 -> .... connected to server
17:06:19.463 -> Body:
17:06:19.463 -> 1eb
17:06:19.463 -> {
17:06:19.463 ->   "kind": "youtube#channelListResponse",
17:06:19.463 ->   "etag": "RTeYeWoPiruQYgKMNGG_kd3IarM",
17:06:19.463 ->   "pageInfo": {
17:06:19.463 ->     "totalResults": 1,
17:06:19.463 ->     "resultsPerPage": 1
17:06:19.463 ->   },
17:06:19.463 ->   "items": [
17:06:19.463 ->     {
17:06:19.463 ->       "kind": "youtube#channel",
17:06:19.463 ->       "etag": "EJ5vRl7NExNelQlFhUk3HEOhLfE",
17:06:19.510 ->       "id": "UCh7RTc1g64bJLwbzP8HXmMw",
17:06:19.510 ->       "statistics": {
17:06:19.510 ->         "viewCount": "73345",
17:06:19.510 ->         "commentCount": "0",
17:06:19.510 ->         "subscriberCount": "842",
17:06:19.510 ->         "hiddenSubscriberCount": false,
17:06:19.510 ->         "videoCount": "49"
17:06:19.510 ->       }
17:06:19.510 ->     }
17:06:19.510 ->   ]
17:06:19.510 -> }
17:06:19.510 -> 
17:06:19.510 -> 0
17:06:19.510 -> 
17:06:19.510 -> 
17:06:19.510 -> END
17:06:19.510 -> Closing client
`

dchadev, 2020-05-15:
Same as above for me.

With debug enabled, in the serial monitor shows that it fetches correctly the data but it doesn't send anything to the displays.

Everything was working just fine up until a couple of days ago.

For reference, I was using this library with the sketch here:
https://www.instructables.com/id/YouTube-Subscriber-Counter-With-ESP8266-V2/

kuuriisuu, 2020-05-15:
Same as @dchadev, I have been using a modified version of the sketch at https://www.instructables.com/id/YouTube-Subscriber-Counter-With-ESP8266-V2/ since September 2019, with the same issue starting at the same time.

I can see a relatively large spike in API traffic before it crashed. Not particularly useful as a clue, but interesting nonetheless.
[image]

FusionSource, 2020-05-17:
I am having the same issue, it was working fine, then just stopped a few days ago, been trying to figure it out, the only thing I have thought of is in the code for the ESP8266 board, you have to set the client to  client.setInsecure(); this might have something to do with it.  I have code written in c# for my mobile phone and this still works.

dchadev, 2020-05-17:
@FusionSource  I tried that the other way and it didn't solve the problem with the sketch I was using. It's weird because the library correctly gets the data from youtube and the backpack library works just fine showing on the display whatever else I throw at them, but it's like if these two parts of the code don't communicate with each other anymore.

InterlinkKnight, 2020-05-17:
So apparently everybody is having the same problem. I'm using it with the ESP32 without the  client.setInsecure(); and still got the same problem.

You think the problem could be the ArduinoJson? We all had to use the version 5 because the version 6 didn't work.

dchadev, 2020-05-17:
@InterlinkKnight That would be my (uneducated) guess, after the tests I've made so far. But I might be wrong as my knowledge about the ArduinoJson library pretty much equals zero for further investigation into that.

InterlinkKnight, 2020-05-19:
Someone found the solution: https://www.youtube.com/watch?v=f4YXnHfK6ZM

I don't understand why it happened, but apparently it works. I say "apparently" because I decided to skip that library all together, for several reasons:
- The library doesn't work with json version 6.
- The library uses "long" integer which limits the amount of views/subscribers that it can show. Famous YouTubers have a lot more than 2,147,483,647 views.
- The library is over-complicated for doing something very simple. Maybe that's why it stopped working without warning.
- I tested this library for several months and sometimes it failed to get the data, so usually I had to try more than once to get the API data.

So my solution was to get the youtube API directly in the sketch, which solved all the problems I mentioned. I'll share my code later on my youtube channel.

witnessmenow, 2020-05-20:
This had nothing to do with ArduinoJson V5 or V6, you can see in the @AntoniousUA serial output , in the body, there is a `1eb` before the JSON starts that should not be there, the body is passed into Arduino JSON and with the 1eb before it was not valid JSON and it failed to parse.

I'm not sure what changed on YouTube/Googles end to make that happen.

This is fixed in #23, this is basically a complete re-write of the library to be way more memory efficient and to use ArduinoJsonV6

PR #23 (2020-05-20): V2
- Code should be much more memory efficient as it does not rely on Arduino String now, although some methods that used String have been left in to retain backwards compatibility.
- Fixed #19 , where the API was returning random characters between the headers and the body
- Fixed #22 , Upgraded to use ArduinoJson V6 - Thanks to #22 and #16 for the PRs, but I ended up completely changing the library so it didn't make sense to bring in your code.
- Added code in the ESP8266 example that allows it to run on ESP8266 core 2.5 and higher (probably #14)
- Added "depends" keyword into library proper [... cut]
closed 2020-05-20 by witnessmenow