witnessmenow/arduino-youtube-api #24: not able to parse correctly

elan22, 2020-05-22:
Hi,
Not sure if this is just me seeing this but I could not get the getChannelStatistics to give back results and after digging ended finding out that for some reason the response string being return had some chars (1f3) right at the beginning and that seemed to mess with the parser.
doing this:
String response = sendGetToYoutube(command);       //recieve reply from youtube
response.remove(0,3);

seemed to solve the issue.

witnessmenow, 2020-05-22:
This was fixed in #23 which was released yesterday.

You would have to update your YouTube library (to v2.0) and Arduino JSON
library (to v6.x)

V2 of this library should be much more memory efficient and probably faster

elan22, 2020-05-22:
aah, I was struggling with this for while now and didn't even check for updates.
Thanks for your work

closed 2020-05-22 by elan22