witnessmenow/arduino-youtube-api #25: Connection failed Unexpected HTTP Status Code: 0

uiaadfc, 2020-05-23:
Hi there, 

Either I'm having problems with the wrapper or im dumber than a box of rocks, I'll try to cut to the chase as quickly as possible. 

I built a subscriber counter using https://www.instructables.com/id/YouTube-Subscriber-Counter-With-ESP8266-V2/ about a year and half ago and its been working fine until this week. 

After seeing everything going on and with other peoples stopped working, saw the wrapper had been updated and updated the wrapper and json to 6.15.2 and now i just get 

```
Connection failed
Unexpected HTTP Status Code: 0

```
In the serial monitor and I've tested with the example code too with the same result.

Any help here would be appreciated

uiaadfc, 2020-05-23:
my own idiocy knows no bounds - forgot to roll the board manager back to below 2.5

closed 2020-05-23 by uiaadfc

witnessmenow, 2020-05-23:
Glad to hear you got sorted. You can use 2.5 or above if you add the line

"client.setInsecure();" to your setup.

I updated the example that comes with the library as part of the update
this week to show to show the change.

MKme, 2020-06-03:
Ran in to this myself today. client.setInsecure(); fixed me up thanks!