witnessmenow/arduino-youtube-api #27: Connection failed

glsf91, 2020-06-25:
My counter (Nodemcu esp8266) was working for years. Suddenly it stopped working.
I saw the closed issues about this.

I updated to your newest version and also ArduinoJson. I use ESP 8266 version 2.4.2.

But now I get:
/youtube/v3/channels?part=statistics&id= removed
Connection failed
Unexpected HTTP Status Code: 0
/youtube/v3/channels?part=statistics&id= removed
Tossing an unexpected character: 1
Tossing an unexpected character: d
Tossing an unexpected character: 1
Tossing an unexpected character: 

Tossing an unexpected character: 

Unexpected HTTP Status Code: 403
Closing client
/youtube/v3/channels?part=statistics&id= removed
Tossing an unexpected character: 1
Tossing an unexpected character: d
Tossing an unexpected character: 1
Tossing an unexpected character: 

and so on.
What can be the problem?

witnessmenow, 2020-06-25:
403 is usually an authentication issue. I would double check your API token
is still valid.

glsf91, 2020-06-25:
I'am sorry. Very stupid.
When trying in a browser I get:
The request cannot be completed because you have exceeded your <a href=\"/youtube/v3/getting-started#quota\">quota

Probably because of the lot of failures.

closed 2020-06-26 by glsf91