witnessmenow/arduino-youtube-api #30: License Change

dmadison, 2020-10-01:
Hi Brian, would you consider changing the license for this library? At the moment it's licensed under GPL v2.0, which I believe means projects that use the library cannot be distributed unless they are open source and also licensed under GPL v2.0.

I would recommend switching to the much more permissive [MIT license](https://opensource.org/licenses/MIT), or if you'd prefer to keep the copyleft provisions then I would recommend the [Lesser General Public License (LGPL)](https://opensource.org/licenses/LGPL-3.0). There should be no issue changing the license to either of those two, as both are more permissive than the current license.

witnessmenow, 2020-10-01:
Hey, I'll take a proper look at changes later but I'm fine for the library
to update to MIT

Brian

dmadison, 2020-10-02:
Thanks for the quick response Brian. I'll try to put together a pull request with the license change in the next day or so.

witnessmenow, 2020-10-02:
Fixed in #33 , thanks

closed 2020-10-02 by witnessmenow