witnessmenow/arduino-youtube-api #40: Suddently returning HTTP 0

MrDesjardins, 2020-11-24:
Hello,
I am using the Arduino Youtube API for about 3 weeks. It worked flawlessly.

However, this morning, the response for the HTTP request is HTTP 0. 

I have verified and the ESP8266 is connecting to the local network (it has an IP).

I have verified and I can perform a HTTP request with the browser with my channel id and API key without a problem.

Is there something I can do?

nirujanr, 2020-11-26:
i fix the issue by update by update youtubeAPI.cpp
`// Get from https://arduinojson.org/v6/assistant/
    const size_t bufferSize = JSON_ARRAY_SIZE(1) + JSON_OBJECT_SIZE(2) + 3*JSON_OBJECT_SIZE(4) + 300;

`
[arduino-youtube-api-2.0.0.zip](https://github.com/witnessmenow/arduino-youtube-api/files/5600369/arduino-youtube-api-2.0.0.zip)

Mafergu16, 2020-12-21:
I'm having the same issue. I tried to update the bufferSize as suggested, but I'm still returning HTTP 0.

dmadison, 2020-12-22:
I can't replicate that error. I've tried both the latest version (V2.0.0) and the latest commit (01ff0a6) with the stock example, just substituting my WiFi info and API key. I've tried with Brian's channel, my channel, and Simone Giertz's channel without issue. I'm using the latest version of ArduinoJSON, 6.17.2.

A few questions for those with issues:
* What channel ID are you using?
* What board and board settings are you using?

nirujanr, 2020-12-22:
- What channel ID are you using? Iam using my own Channel ID
   - What board and board settings are you using? esp8266

[image: image.png]

nirujanr, 2020-12-22:
//    Arduino IDE        v1.8.9
//    esp8266 board file v2.4.2
//    YoutubeAPI         v1.1.0
//    ArduinoJson        v5.13.4
//    LedControl         v1.0.6
//

dmadison, 2020-12-22:
Have you considered updating your software? You're using out of date versions for this library, ArduinoJSON, and the ESP8266 boards. As I said I could not replicate your issue using the latest releases.

Mafergu16, 2020-12-22:
> Have you considered updating your software? You're using out of date versions for this library, ArduinoJSON, and the ESP8266 boards. As I said I could not replicate your issue using the latest releases.

I'm actually using the following versions:

// Tested using:
//    Arduino IDE        v1.8.9
//    esp8266 board file v2.7.4
//    YoutubeAPI         v2.0.0
//    ArduinoJson        v6.17.2
//    LedControl         v1.0.6

Yeah, I'm not sure why. Maybe it's the code I'm using from here: 

https://github.com/GadgetReboot/Arduino/blob/master/ESP8266/YoutubeSubCounter_7Seg_WEMOS/Sub_Counter_7seg.ino

dmadison, 2020-12-22:
Have you tested with the library examples instead of that program? If the library examples work then the problem likely lies outside of the library.

At a glance I notice that that code is missing this line from the setup function:

```cpp
    client.setInsecure();
```

This is required with [ESP8266 core v2.5.0](https://github.com/esp8266/Arduino/releases/tag/2.5.0) and above, and its absence would cause the client to report a status code of 0.

Mafergu16, 2020-12-23:
> ```c++
> client.setInsecure();
> ```

Well, that did it! Thank you so much for your help!

witnessmenow, 2021-07-07:
Library is currently working when following the examples so I dont believe this is an issue

closed 2021-07-07 by witnessmenow