witnessmenow/arduino-youtube-api #41: NOT Changing Subscriber Count: 1000

secginsecgin, 2021-01-16:
Hi,
After 1000 subscriber it has stopped to count. other things are working. please find the serial port screen output
---------Stats---------
Subscriber Count: 1000 - till 1000 it worked
View Count: 213410 - this is correct and live

dmadison, 2021-01-16:
How many subscribers do you have, exactly?

On September 10th, 2019 [Google changed the API so that subscriber counts are rounded to three significant figures](https://developers.google.com/youtube/v3/revision_history#release_notes_09_10_2019), so you'll lose the last digit of precision.

Brian himself made a nice video about the change which you can find here:
* [YouTube Subscriber counters are about to break (so I made a lot of them)](https://www.youtube.com/watch?v=sHNI-WgN-UQ)

witnessmenow, 2021-07-06:
Not an issue, congrats on being too big for an accurate counter :)

closed 2021-07-06 by witnessmenow