witnessmenow/arduino-youtube-api #43: how can i use the subscriber number for custom alert (string)

MagmaBeast, 2021-05-05:
is the subscriber in form of a string , char or integer so i can use basic statement such as
if(api.channelStats.subscriberCount = '2' ){
    digitalWrite(2,HIGH);}
    else{
      digitalWrite(2,LOW);
    }

joeybab3, 2021-05-05:
according ot the source code its a `long`: https://github.com/witnessmenow/arduino-youtube-api/blob/01ff0a667f64e5ae47456a2e90f98a31858eee5b/src/YoutubeApi.h#L43

witnessmenow, 2021-05-05:
> According to the source code it's a long:

`== 2` should work

MagmaBeast, 2021-05-06:
Thank u it worked

MagmaBeast, 2021-05-10:
can u suggest me a way so that everytime someone subscribes a buzzer buzzes for 10 seconds .thanks

closed 2021-05-10 by MagmaBeast

MagmaBeast, 2021-05-11:
here is the code  os far doesnt work :(

/*******************************************************************
    Read YouTube Channel statistics from the YouTube API on
    an ESP8266 and print them to the serial monitor
                                                                  
    Parts:
    D1 Mini ESP8266 (or any ESP8266) * - http://s.click.aliexpress.com/e/uzFUnIe
    * = Affilate
   
    If you find what I do useful and would like to support me,
    please consider becoming a sponsor on Github
    https://github.com/sponsors/witnessmenow/

    Written by Brian Lough
    YouTube: https://www.youtube.com/brianlough
    Tindie: https://www.tindie.com/stores/brianlough/
    Twitter: https://twitter.com/witnessmenow
 *******************************************************************/
 #include <Servo.h>
Servo servo; 

// ---------------------------
// Standard Libraries
// ----------------------------

#include <ESP8266WiFi.h>
#include <WiFiClientSecure.h>

// ----------------------------
// Additional Libraries - each one of these will need to be installed.
// ----------------------------

#include <YoutubeApi.h>
// Library for connecting to the Youtube API

// Search for "youtube" in the Arduino Library Manager
// https://github.com/witnessmenow/arduino-youtube-api

#include <ArduinoJson.h>
// Library used for parsing Json from the API responses

// Search for "Arduino Json" in the Arduino Library manager
// https://github.com/bblanchon/ArduinoJson

//------- Replace the following! ------
char ssid[] = "realme 7";       // your network SSID (name)
char password[] = "5ada945d25df";  // your network key
#define API_KEY "AIzaSyCEZY8XhpjJLURxbl7CVdB5RQcw4Yfwo8U"  // your google apps API Token
#define CHANNEL_ID "UC34fGHsu-MXw3-S9FxaZqEw" // makes up the url of channel
//------- ---------------------- ------

WiFiClientSecure client;
YoutubeApi api(API_KEY, client);

unsigned long timeBetweenRequests = 0;
unsigned long nextRunTime;

long subs = 0;

void setup() {
  
  servo.attach(D4);
servo.write(90);
  Serial.begin(115200);
digitalWrite(D2,LOW);
  // Set WiFi to station mode and disconnect from an AP if it was Previously
  // connected
  WiFi.mode(WIFI_STA);
  WiFi.disconnect();
  delay(0);

  // Attempt to connect to Wifi network:
  Serial.print("Connecting Wifi: ");
  Serial.println(ssid);
  WiFi.begin(ssid, password);
  while (WiFi.status() != WL_CONNECTED) {
    Serial.print(".");
    delay(0);
  }
  Serial.println("");
  Serial.println("WiFi connected");
  Serial.println("IP address: ");
  IPAddress ip = WiFi.localIP();
  Serial.println(ip);

  // Required if you are using ESP8266 V2.5 or above
  client.setInsecure();

  // If you want to enable some extra debugging
  api._debug = false;
}
 
void loop() {

  if (millis() > nextRunTime)  {
    if(api.getChannelStatistics(CHANNEL_ID))
    {
     
      Serial.print("Subscriber Count: ");
      Serial.println(api.channelStats.subscriberCount);
     subs = api.channelStats.subscriberCount +1 ;

    }
    nextRunTime = millis() + timeBetweenRequests;
  }
  
  if(subs == api.channelStats.subscriberCount + 1 ){
    digitalWrite(D5,HIGH);
    delay(10000);
    digitalWrite(D5,LOW);
    }
 

}