witnessmenow/arduino-youtube-api #45: Error when compiling (warning: deprecated conversion & redeclaration)

r4ph74, 2021-07-06:
hi I'm still new in this Arduino world,  I used to work with this project, and works fine. no problem at all. 

It's been quite a long time I don't get this project again until today I got this project and I tried to compile it and comes up with this error.  I haven't change the code and it works fine before .. just replace the wifi id * pass and channel id. 

below is the error message  from the Arduino console:

In file included from F:\ESP32\Project\YT_IG_Counter_Fix\YT_IG_Counter_WIFI\YT_IG_Counter_WIFI.ino:19:0:
C:\Users\r4ph\Documents\Arduino\libraries\YoutubeApi-2.0.0\src/YoutubeApi.h:54:23: error: redeclaration of 'channelStatistics YoutubeApi::channelStats'
     channelStatistics channelStats;
                       ^
C:\Users\r4ph\Documents\Arduino\libraries\YoutubeApi-2.0.0\src/YoutubeApi.h:50:23: note: previous declaration 'channelStatistics YoutubeApi::channelStats'
     channelStatistics channelStats;
                       ^
F:\ESP32\Project\YT_IG_Counter_Fix\YT_IG_Counter_WIFI\YT_IG_Counter_WIFI.ino:50:31: warning: deprecated conversion from string constant to 'char*' [-Wwrite-strings]
 YoutubeApi api(API_KEY, client);
                               ^
F:\ESP32\Project\YT_IG_Counter_Fix\YT_IG_Counter_WIFI\YT_IG_Counter_WIFI.ino: In function 'void loop()':
F:\ESP32\Project\YT_IG_Counter_Fix\YT_IG_Counter_WIFI\YT_IG_Counter_WIFI.ino:109:46: warning: deprecated conversion from string constant to 'char*' [-Wwrite-strings]
       if (api.getChannelStatistics(CHANNEL_ID))
                                              ^
exit status 1
Error compiling for board LOLIN(WEMOS) D1 R2 & mini.
In file included from F:\ESP32\Project\YT_IG_Counter_Fix\YT_IG_Counter_WIFI\YT_IG_Counter_WIFI.ino:19:0:
C:\Users\r4ph\Documents\Arduino\libraries\YoutubeApi-2.0.0\src/YoutubeApi.h:54:23: error: redeclaration of 'channelStatistics YoutubeApi::channelStats'
     channelStatistics channelStats;
                       ^
C:\Users\r4ph\Documents\Arduino\libraries\YoutubeApi-2.0.0\src/YoutubeApi.h:50:23: note: previous declaration 'channelStatistics YoutubeApi::channelStats'
     channelStatistics channelStats;
                       ^
F:\ESP32\Project\YT_IG_Counter_Fix\YT_IG_Counter_WIFI\YT_IG_Counter_WIFI.ino:50:31: warning: deprecated conversion from string constant to 'char*' [-Wwrite-strings]
 YoutubeApi api(API_KEY, client);
                               ^
F:\ESP32\Project\YT_IG_Counter_Fix\YT_IG_Counter_WIFI\YT_IG_Counter_WIFI.ino: In function 'void loop()':
F:\ESP32\Project\YT_IG_Counter_Fix\YT_IG_Counter_WIFI\YT_IG_Counter_WIFI.ino:109:46: warning: deprecated conversion from string constant to 'char*' [-Wwrite-strings]
       if (api.getChannelStatistics(CHANNEL_ID))
                                              ^
exit status 1
Error compiling for board LOLIN(WEMOS) D1 R2 & mini.

================================

I hope some can help me how to resolve this error. 

Thanks in Advance

Rafael 

I'm using :
Wemos D1 mini 
Arduino  1.8.13
YoutubeAPI :  V2
ArduinoJson : 6.18.0

dmadison, 2021-07-06:
It looks as though you may have a duplicate line in your file. I can't replicate that error here. Try compiling one of the examples to see if you have the same error. If so try downloading a fresh copy of the library. If not, you'll need to share your code to look into the issue further.

The `deprecated conversion` warning is a known issue. There is a pull request open to fix it (#34).

r4ph74, 2021-07-06:
I just tried to compile the original example from the library still gives me the error. 

---

In file included from C:\Users\r4ph\Documents\Arduino\libraries\YoutubeApi-2.0.0\examples\ESP8266\ChannelStatistics\ChannelStatistics.ino:30:0:
C:\Users\r4ph\Documents\Arduino\libraries\YoutubeApi-2.0.0\src/YoutubeApi.h:54:23: error: redeclaration of 'channelStatistics YoutubeApi::channelStats'
     channelStatistics channelStats;
                       ^
C:\Users\r4ph\Documents\Arduino\libraries\YoutubeApi-2.0.0\src/YoutubeApi.h:50:23: note: previous declaration 'channelStatistics YoutubeApi::channelStats'
     channelStatistics channelStats;
                       ^
C:\Users\r4ph\Documents\Arduino\libraries\YoutubeApi-2.0.0\examples\ESP8266\ChannelStatistics\ChannelStatistics.ino:50:31: warning: deprecated conversion from string constant to 'char*' [-Wwrite-strings]
 YoutubeApi api(API_KEY, client);
                               ^
C:\Users\r4ph\Documents\Arduino\libraries\YoutubeApi-2.0.0\examples\ESP8266\ChannelStatistics\ChannelStatistics.ino: In function 'void loop()':
C:\Users\r4ph\Documents\Arduino\libraries\YoutubeApi-2.0.0\examples\ESP8266\ChannelStatistics\ChannelStatistics.ino:91:43: warning: deprecated conversion from string constant to 'char*' [-Wwrite-strings]
     if(api.getChannelStatistics(CHANNEL_ID))
                                           ^
exit status 1
Error compiling for board LOLIN(WEMOS) D1 R2 & mini.
----------

this is the original sketch example file that I took from the library. 
[ChannelStatistics.zip](https://github.com/witnessmenow/arduino-youtube-api/files/6771949/ChannelStatistics.zip)

And this is the youtubeAPI.h code . (the original code) 

/*
Copyright (c) 2020 Brian Lough. All right reserved.

YoutubeApi - An Arduino wrapper for the YouTube API

This library is free software; you can redistribute it and/or
modify it under the terms of the GNU Lesser General Public
License as published by the Free Software Foundation; either
version 2.1 of the License, or (at your option) any later version.

This library is distributed in the hope that it will be useful,
but WITHOUT ANY WARRANTY; without even the implied warranty of
MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE. See the GNU
Lesser General Public License for more details.

You should have received a copy of the GNU Lesser General Public
License along with this library; if not, write to the Free Software
Foundation, Inc., 51 Franklin St, Fifth Floor, Boston, MA 02110-1301 USA
*/

#ifndef YoutubeApi_h
#define YoutubeApi_h

#include <Arduino.h>
#include <ArduinoJson.h>
#include <Client.h>

#define YTAPI_HOST "www.googleapis.com"
#define YTAPI_SSL_PORT 443
#define YTAPI_TIMEOUT 1500

#define YTAPI_CHANNEL_ENDPOINT "/youtube/v3/channels"

struct channelStatistics{
  long viewCount;
  long commentCount;
  long subscriberCount;
  bool hiddenSubscriberCount;
  long videoCount;
};

class YoutubeApi
{
  public:
    YoutubeApi (char *apiKey, Client &client);
    YoutubeApi (String apiKey, Client &client);
    int sendGetToYoutube(char *command);
    bool getChannelStatistics(char *channelId);
    bool getChannelStatistics(String channelId);
    channelStatistics channelStats;
    bool _debug = false;
	 String getMyRecentSubscribers();
    String getMyRecentSubscribers(String pageToken);
    channelStatistics channelStats;
    String myRecentSubscribers[5]; // Fixed number for now, shouldn't go above 50

  private:
    char *_apiKey;
    Client *client;
    int getHttpStatusCode();
    void skipHeaders();
    void closeClient();
};

#endif
===============================================================

I think there might be some changes in the parameter of youtubeApi.  but I don't know how to fix it.

dmadison, 2021-07-06:
That header has been modified. The `channelStatistics` member has been declared twice.

r4ph74, 2021-07-07:
I see.

Should I just re-install or just copy paste the original header to mine?

r4ph74, 2021-07-07:
I just reinstall the library again. 

the re declaration is already gone .. but there are some warnings.. regarding string constant as below 

=============
F:\ESP32\Project\test\ChannelStatistics\ChannelStatistics.ino:50:31: warning: deprecated conversion from string constant to 'char*' [-Wwrite-strings]
 YoutubeApi api(API_KEY, client);
                               ^
F:\ESP32\Project\test\ChannelStatistics\ChannelStatistics.ino: In function 'void loop()':
F:\ESP32\Project\test\ChannelStatistics\ChannelStatistics.ino:91:43: warning: deprecated conversion from string constant to 'char*' [-Wwrite-strings]
     if(api.getChannelStatistics(CHANNEL_ID))
                                           ^
Executable segment sizes:
IROM   : 357032          - code in flash         (default or ICACHE_FLASH_ATTR) 
IRAM   : 27184   / 32768 - code in IRAM          (ICACHE_RAM_ATTR, ISRs...) 
DATA   : 1288  )         - initialized variables (global, static) in RAM/HEAP 
RODATA : 1756  ) / 81920 - constants             (global, static) in RAM/HEAP 
BSS    : 25224 )         - zeroed variables      (global, static) in RAM/HEAP 
Sketch uses 387260 bytes (37%) of program storage space. Maximum is 1044464 bytes.
Global variables use 28268 bytes (34%) of dynamic memory, leaving 53652 bytes for local variables. Maximum is 81920 bytes.
=================================

i

witnessmenow, 2021-07-07:
As @dmadison points out, this is a modified code that didn't come with the library so it is not an issue with the library. Delete and re-install the library if you want to get back to a working version

closed 2021-07-07 by witnessmenow

r4ph74, 2021-07-07:
Thanks @witnessmenow and @witnessmenow for your support.