witnessmenow/arduino-youtube-api #49: example code leads to an HTTP status 0

taewyth, 2021-11-13:
I wanted to try the code example to make sure everything worked, but it returns and HTTP status 0, so I added the ESP32Ping library to ping googleapis.com to see if at least it could be reached, and it apparently does, that's the only modification made to the example code (aside from the necessary ones)

Here's the serial monitor readings:
[image]

I also tried with your channel ID (as provided on the readme file), the issue stays here.

for more details:
* YoutubeApi version: 2.0.0
* ArduinoJson version: 6.18.5
* [card used](https://joy-it.net/en/products/SBC-NodeMCU-ESP32)

witnessmenow, 2021-11-13:
I think its a HTTPs verification thing. ESP32 changed the default behavior not too log ago 

Change this:

```
#ifdef ESP8266
// Required if you are using ESP8266 V2.5 or above
client.setInsecure();
#endif
```

to 

```
client.setInsecure();
```

And it should work.

taewyth, 2021-11-14:
Yep, that was it.

closed 2021-11-14 by taewyth